adafruit/Adafruit_CircuitPython_CharLCD #1: another impl (not an issue, just a pointer)

dhalbert, 2017-07-05:
FYI, you may be interested in https://github.com/dhalbert/CircuitPython_LCD, for char LCDs, with the interface factored out. Currently the only interface I provide is for PCF8574 backpacks (not compatible with Adafruit backpack). I did some things to reduce RAM usage as well (const(), underscored constant names, etc.)

brentru, 2017-07-05:
Thanks for linking your implementation! (it's cool to see other's implementations). I'll look into using these methods for reducing RAM usage as well.

closed 2017-07-05 by brentru

dhalbert, 2017-07-05:
https://docs.micropython.org/en/latest/esp8266/reference/constrained.html#execution-phase talks about `const()`, etc.. And see other stuff on that page as well.