adafruit/Adafruit_CircuitPython_CharLCD #5: Character_LCD set_backlight function is backwards

BravoDelta151, 2017-12-12:
If you pass True, the value is set to 0 (False) turning the backlight off and you must pass False to turn it on.  

Additionally, a quick review of the __init__ function leaves the self.backlight member set to None which will cause an exception if you attempt to call set_backlight since None has no 'value' member.

def set_backlight(self, lighton):
      """ Set lighton to turn the charLCD backlight on.
            :param lighton: True to turn backlight on, False to turn off
      """
      if lighton:
        self.backlight.value = 0
      else:
        self.backlight.value = 1

tdicola, 2017-12-15:
Yeah I noticed this too and it's not necessarily wrong but rather assuming a common cathode backlight (like from some RGB displays).  I'm not familiar with all the types of LCD backlights to make the call on the right way, the Pi version of this library actually lets you specify if the backlight logic is inverted or not (vs. this library that assumes it's inverted).  I think ultimately this library might want the same concept of inverted vs. non-inverted backlights (i.e. is backlight hooked up to the LED anode or cathode).

TurbulentCalm, 2018-01-17:
Agreed, all my monochrome LCD require this to be set to False to enable the backlight.

ladyada, 2018-11-21:
@kattni this is resolved now?

kattni, 2018-11-21:
Yep this is fixed in #19

closed 2018-11-21 by kattni