adafruit/Adafruit_CircuitPython_CharLCD #21: Feature request - Begin message at cursor position

kattni, 2018-11-21:
Currently, when you call `lcd.message` it begins the message at the beginning of the line. This is by design.

There was an issue filed on the Python library requesting the ability to "disable cursor shift". The desired result is that the cursor stay at the position set by `cursor_position` when `message` is called.

One option we can consider is adding another version of `message` that does not reset to the beginning of the line for each new message.

ladyada, 2018-12-21:
you can subscibe to this issue to be notified when it'll be added, but its jsut a placeholder for now, we don't have an ETA when we'll get to it. if you figure out how, submit a pull request and we'll check it out :)

tannewt, 2019-01-03:
@n0xy1 Would Python string formatting help? https://pyformat.info/

TheHWcave, 2019-01-23:
I definitely support having a 2nd message function that doesn't touch the current cursor position. Alternatively an additional boolean parameter in the original function could be used to turn the current behaviour off. That parameter could be defaulted to the current behaviour so no existing code would break.  
in the mean time and in response to n0xy1, here is my work-around. Just define our own message function like that  (yeah, i know one isn't supposed to call "_names" but in a crunch what can you do)?

def Kludge(col,row,txt):
	lcd.cursor_position(col,row)
	for c in txt:
		lcd._write8(ord(c),True)

use like this to write the string "hello" starting from the 4th char position on row 2 without touching anything else already displayed:

Kludge(3,1,'hello')

profbrady, 2019-04-05:
Any movement on this? Last year for a class I run in the spring I was able to have students use the library to display text on two lines of a 16x2 lcd at any position by using `.set_cursor()`. Now that it is `.cursor_position()` it does not work. It seems that the `message` command ignores that position set by `.cursor_position()`.

profbrady, 2019-04-06:
I just created a request to make a change that seems to fix this issue. I tested it with my assignment from last year and it worked.

profbrady, 2019-04-12:
My request was just approved. It is referred to as column_align, but it also makes cursor_position work for the next message before resetting to 0,0.

makermelissa, 2019-04-12:
Ok, I changed the release text. Thanks again.

makermelissa, 2019-04-12:
I just updated the release notes to include that @profbrady.

kattni, 2019-04-12:
Fixed by #37

closed 2019-04-12 by kattni