adafruit/Adafruit_CircuitPython_CharLCD #32: Unresolved import

IgnorantGeek, 2019-02-28:
When trying to import this package with

`import adafruit_character_lcd.character_lcd as characterlcd`

I receive a pylint error saying unresolved import. I have install the package with the command 

`sudo pip3 install adafruit-circuitpython-charlcd`

Not sure where to go from here. I would like to use an LCD with my RPi so I was hoping to make use of this package. I am using VSCode and Python 3.6.

dhalbert, 2019-02-28:
This would seem to be just a pylint issue; it may not be finding the module. I replicated your setup, more or less, and didn't get an error when I actually imported the module. Note that `sudo pip3` is not necessary. `pip3` will install in `~/.local/lib`, with modern versions of pip3. That's safer, and better unless you're trying to install for all users on the RPi.

```
pi@raspberrypi:~ $ pip3 install adafruit-circuitpython-charlcd
Collecting adafruit-circuitpython-charlcd
[...]
Successfully installed Adafruit-Blinka-1.2.4 Adafruit-PlatformDetect-0.0.10 Adafruit-PureIO-0.2.3 RPi.GPIO-0.6.5 adafruit-circuitpython-74hc595-1.0.2 adafruit-circuitpython-busdevice-2.2.9 adafruit-circuitpython-charlcd-3.1.2 adafruit-circuitpython-mcp230xx-1.2.0 rpi-ws281x-4.1.0 spidev-3.4 sysv-ipc-1.0.0
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_character_lcd.character_lcd as characterlcd
>>> 
```

IgnorantGeek, 2019-02-28:
Ok, I also tried running the pip3 install statement without the sudo flag and it seems to have installed correctly, so I am not sure why pylint isn't picking it up. Could it be that I am not actually programming on the Pi? I was going to write the code on my desktop and then upload it to my Pi to run it there. Not sure how the libraries are setup or how the dependencies work but maybe the library doesn't like that it can't detect the actual pins? Doesn't seem like that's how it should work.

IgnorantGeek, 2019-02-28:
I was able to get the library to load properly on another machine so it seems like it's probably a setting of mine. I'm going to go ahead and close this issue.

closed 2019-02-28 by IgnorantGeek

dhalbert, 2019-02-28:
You do have to run `pip3` on the RPi to install the library, because it's going to run on that machine, not on your editor desktop. Since the RPi is a computer in its own right, and you may want to run an editor on it. You could `pip3 install` on your desktop, but that would just be to make pylint happy.