adafruit/Adafruit_CircuitPython_CharLCD #34: Backlight not turn on

wiryonolau, 2019-03-19:
Hi I'm using this product
https://ae01.alicdn.com/kf/HTB1ZBj9n4PI8KJjSspfq6ACFXXay/16x2-LCD-Display-Shield-Module-RGB-backlight-for-Raspberry-Pi-3-Model-B-2B.jpg_q50.jpg

Test using i2c_rgb from your example script, the text appear but the back light is not turn on.
In the top part of the lcd there is small led which on changing led.color parameter that led is change color. but backlight is dead

Any solution for this ?

ladyada, 2019-03-19:
hiya - you will have to chat with the vendor for tech support - we don't own this device, it could have the LEDs set up differently

closed 2019-03-19 by ladyada

wiryonolau, 2019-03-19:
Yeah they wired differently and didn't give any manual xD

When I use RPLCD the backlight is lit on , but the LED above LCD is not and text is not appear
But When I use Adafruit_CircuitPython_CharLCD, the text appear but backlight is not lit , LED has RGB color.

Guess I need to combine both of this

ranguli, 2019-06-15:
@wiryonolau Hi there, did you ever figure out how to get this working? I have the exact same product, and am also not able to get the backlight working.

With the example code below, the text is displayed and the status LED lights up, but the backlight does _not_ light up.

```python
import board
import time
import busio
import adafruit_character_lcd.character_lcd_rgb_i2c as character_lcd

lcd_columns = 16
lcd_rows = 2

i2c = busio.I2C(board.SCL, board.SDA)
lcd = character_lcd.Character_LCD_RGB_I2C(i2c, lcd_columns, lcd_rows)

lcd.backlight = True
lcd.color = [100, 0, 0]
lcd.message = "Hello world"
```

wiryonolau, 2019-06-15:
> @wiryonolau Hi there, did you ever figure out how to get this working? I have the exact same product, and am also not able to get the backlight working.
>
> With the example code below, the text is [...]

Yeah, make sure it is sainsmart
```bash
# import set_bit
from adafruit_character_lcd.character_lcd import _setbit as set_bit

...
lcd = character_lcd.Character_LCD_RGB_I2C(i2c, lcd_columns, lcd_rows)
lcd._mcp.iodira = set_bit(lcd._mcp.iodira, 5, 0)

# Turn on light
lcd._mcp.gpioa = set_bit(lcd._mcp.gpioa, 5, 0)

# Turn off light
lcd._mcp.gpioa = set_bit(lcd._mcp.gpioa, 5, 1)
```
When turning the backlight on it will set the led above to different color, just change it after the backlight on if you are bother by it :D