adafruit/Adafruit_CircuitPython_CharLCD #36: MemoryError on Metro M0 Express

caternuson, 2019-03-29:
Re:
https://forums.adafruit.com/viewtopic.php?f=63&t=149682

Can recreate with following:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Metro M0 Express with samd21g18
>>> import board, busio
>>> import adafruit_character_lcd.character_lcd_i2c as character_lcd
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> lcd = character_lcd.Character_LCD_I2C(i2c, 20, 4)
>>> lcd.message = "Hello\nCircuitPython"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_character_lcd/character_lcd.py", line 338, in message
MemoryError: memory allocation failed, allocating 160 bytes
>>> 
```

But seems to work on M4:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Metro M4 Express with samd51j19
>>> import board, busio
>>> import adafruit_character_lcd.character_lcd_i2c as character_lcd
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> lcd = character_lcd.Character_LCD_I2C(i2c, 20, 4)
>>> lcd.message = "Hello\nCircuitPython"
>>> 
```

ladyada, 2019-03-29:
try 4.x - if that is still not fitting then yeah this library could be slimmed down, i see a lot of inefficiencies. let me know if you want to take that on

caternuson, 2019-03-29:
similar with 4.x:
```python
Adafruit CircuitPython 4.0.0-beta.5 on 2019-03-17; Adafruit Metro M0 Express with samd21g18
>>> import board, busio
>>> import adafruit_character_lcd.character_lcd_i2c as character_lcd
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> lcd = character_lcd.Character_LCD_I2C(i2c, 20, 4)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_character_lcd/character_lcd_i2c.py", line 74, in __init__
  File "adafruit_mcp230xx.py", line 31, in <module>
MemoryError: 
>>> 
```

@kattni - you up for this?

ladyada, 2019-03-29:
i think @kattni's pretty booked, this might be better for you to take on

dhalbert, 2019-03-29:
I have a different library that has similar functionality, where the interface is factored out of the user API: https://github.com/dhalbert/CircuitPython_LCD. It was written before this one; they come from different source bases. The current library takes the parallel interface as the base and then adapts other interfaces to that. The `minimal` branch in my library is actually the one I've been using. It removes RTL and a bunch of other lesser-used features to save memory.

caternuson, 2019-05-03:
@dhalbert Question about your library - how would it support usage of bare displays which would require the parallel interface? For example:
https://www.adafruit.com/product/181
and this hardware setup:
https://learn.adafruit.com/character-lcds/python-circuitpython

kattni, 2019-06-03:
Tested on Metro M0. Memory fails with the RGB LCD I2C shield and the I2C RGB simpletest.

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 15, in <module>
  File "adafruit_character_lcd/character_lcd_rgb_i2c.py", line 86, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 34, in <module>
MemoryError: memory allocation failed, allocating 64 bytes
```

Works properly on Metro M4.

caternuson, 2019-06-20:
Ugh...confusing results...
@kattni Can you check again please? Or let me know of any HW differences with your setup.
 
Finally getting back to this. Had to order up the RGB shield kit HW. I've now got the following:
* Metro M0 Express ([PID 3505](https://www.adafruit.com/product/3505))
* RGB LCD Shield Kit ([PID 714](https://www.adafruit.com/product/714))

Using CP 4.0.1 with bundle release from 20190620 and running this example:
https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/blob/master/examples/charlcd_i2c_rgb_simpletest.py
saved as code.py. It seems to work for me:
```python
Adafruit CircuitPython 4.0.1 on 2019-05-22; Adafruit Metro M0 Express with samd21g18
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
```
[image]

kattni, 2020-05-04:
@caternuson I don't have the hardware here to test this here, but can you test it on 5.x? If you don't experience the issue, we'll close it.

caternuson, 2020-05-05:
@kattni :(

```python
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit Metro M0 Express with samd21g18
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 5, in <module>
  File "adafruit_character_lcd/character_lcd_rgb_i2c.py", line 58, in <module>
MemoryError:

Press any key to enter the REPL. Use CTRL-D to reload.
```

kattni, 2020-05-11:
@caternuson Bummer. Alright. You were looking into this last, and couldn't replicate it. It appears that you are now able to. It appears the last thing was you needing more information from Dan. Is that accurate? Is this still something you're willing/able to look into?

caternuson, 2020-05-12:
@kattni I self assigned and will take a look.

pkoppstein, 2020-07-19:
Any progress on this?  I ran into the problem with a Feather M0 Express with libraries from
adafruit-circuitpython-bundle-5.x-mpy-20200709

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit Feather M0 Express with samd21g18
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "adafruit_character_lcd/character_lcd_i2c.py", line 47, in <module>
  File "adafruit_character_lcd/character_lcd.py", line 118, in <module>
MemoryError: memory allocation failed, allocating 96 bytes
>>> 
```

Workarounds for using an LCD with I2C backpack?