adafruit/Adafruit_CircuitPython_CharLCD #50: When using digitalio for RGB control on RPi, list values must be more than 1 to actually turn on corresponding LED

taotien, 2020-08-30:
`pip freeze` 
`Adafruit-Blinka==5.3.2`
`adafruit-circuitpython-charlcd==3.3.3`

## Input
```
import adafruit_character_lcd.character_lcd as characterlcd
import board
import digitalio

lcd_rs = digitalio.DigitalInOut(board.D4)
lcd_en = digitalio.DigitalInOut(board.D19)
lcd_d7 = digitalio.DigitalInOut(board.D27)
lcd_d6 = digitalio.DigitalInOut(board.D22)
lcd_d5 = digitalio.DigitalInOut(board.D24)
lcd_d4 = digitalio.DigitalInOut(board.D25)
lcd_cols = 16
lcd_rows = 2
red = digitalio.DigitalInOut(board.D21)
green = digitalio.DigitalInOut(board.D12)
blue = digitalio.DigitalInOut(board.D18)
lcd = characterlcd.Character_LCD_RGB(
    lcd_rs,
    lcd_en,
    lcd_d4,
    lcd_d5,
    lcd_d6,
    lcd_d7,
    lcd_cols,
    lcd_rows,
    red,
    green,
    blue)

RED = [1, 0, 0]
GREEN = [0, 1, 0]
BLUE = [0, 0, 1]

lcd.color = GREEN
```

## Expected outcome
LCD backlight is bright green

## Actual outcome
LCD backlight stays dark/off

## Current workaround
Set values for `RED, GREEN, BLUE` to anything more than 1, such as 2, or 100. The outcomes are all the same as long as it's >1.
Overall this isn't much of an issue as `pulseio.PWMOut()` has been [patched](https://github.com/adafruit/Adafruit_Blinka/pull/314) already and works.

lesamouraipourpre, 2021-05-14:
With the move from `pulseio.PWMOut` to `pwmio.PWMOut` has this been fixed?

Ref: 6a87186f50eae4cd6da0e85132bbae70226ec275, f21f415149b28eade9169215e0a26155981497f1 and 7e643baa6f64490edf32f89934832a1e69b35f25

jposada202020, 2021-05-14:
@taotien Would you mind retesting this, as lesamouraipourpre I do not have the hardware. Thanks

taotien, 2021-05-17:
Sorry for the confusion, but `pulseio.PWMOut` was a workaround to a problem with `digitalio` (and iirc not recommended in the tutorial), not the issue itself. Unfortunately I have a different hardware issue currently and can't test it either. I would assume that `pwmio.PWMout` functions the same as `pulseio.PWMOut`.

jposada202020, 2021-05-17:
No problem, thanks for letting us know, we will close now, and it could be re-open if necessary if the problem comes back. Thanks.

closed 2021-05-17 by jposada202020