adafruit/Adafruit_CircuitPython_CharLCD #53: Error in docs

jpstroop, 2020-12-06:
This may not be the place, but I can't quickly find the source for the website online. Just beneath [here](https://learn.adafruit.com/character-lcds/python-circuitpython#python-computer-wiring-3006769-6) in the Character LCDs docs, there is an typo regarding the wiring of an LCD with an RGB backlight (second example). It says 

> Pi GND to LCD pin 1, 5, 16, and the opposite side of the potentiometer.

but for the RGB backlit device, LCD pin 16 controls the red backlight.

Presumably is was just pasted from the first example 😄

dherrada, 2020-12-11:
@jpstroop Thanks for catching that. I'll fix it sometime in the next few days

dherrada, 2020-12-16:
Fixed

closed 2020-12-16 by dherrada