adafruit/Adafruit_CircuitPython_CharLCD #55: Message draw slow compared to legacy library (Adafruit_CharLCDPlate)

LamptonWorm, 2021-01-23:
Hi,

I'm using one of the older Adafruit i2c 16x2 RGB LCD Pi Plates with a Rasp. Pi 2 Model B v1.1
CircuitPython_CharLCD is working fine, but message draw is slow compared to the legacy library here [https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/legacy/Adafruit_CharLCDPlate](url).
You can see each letter drawing in turn, whereas the legacy code is almost instant.

Thanks,

Arnwaltur, 2022-07-16:
Same experience here. It makes using a menu rather unpleasant.

szyg, 2022-07-29:
Also noticed the same problem.  The slow refresh makes any updates to the display very jarring compared to the old library.  Any plans to update this library? Any luck by others to increase the draw speed or did you just revert to the old library?

ladyada, 2022-08-01:
are you both using older raspi computers? its a little slower here due to a refactor

szyg, 2022-08-01:
I am running on a Pi 3b model a22082.  New library displays with a very distracting left to right sweep on every redraw where the old is barely a flicker.  Running Raspberry OS 11 Bullseye if that is useful.  Using the converted legacy library is working great for now.

ladyada, 2022-08-01:
@Neradoc do you want to try improving the performance? basically we should write 4 bits at a time rather than using DigitalIO which is more elegant but slower

dhalbert, 2022-08-01:
I wrote my own char LCD library a long time ago for PCF8574 chips: https://github.com/dhalbert/CircuitPython_LCD. It does 4-bit writes. I think it could be adapted to the MCP23008 supported by this library. It is also factored in such a way that the chip-specific logic is separate from the main library.

Arnwaltur, 2022-08-02:
@ladyada Yes, using Pi 1 & 2. It is slow and the CPU load is high compared to the old library. I use it for a web radio, and it's not uncommon for the sound to chop when the screen displays new text.

LamptonWorm, 2022-08-05:
Hi, just to confirm, yes I was also seeing this on an older Pi 2 Model B v1.1 .. thanks!

PaulHutchInMA, 2023-09-14:
I just bought the _I2C/SPI backpack_ (#292) with a _Standard LCD 16x_2 (#181) and am using it with a _MCP2221_ (#4471) on a PC running Debian 11. It takes over 6 seconds to write "Hello CircuitPython" to the display.

I tried it on a Windows 10 PC with the same result.

I found the display update speed does not change when I switch the MCP2221's  I2C speed from 100kHz to 400kHz.

This is almost completely useless, I could even live with 1 second to write 19 characters but >6 seconds is not useful.

ChaoticPumpkin, 2023-09-29:
If there isn't any easy way to fix the data transmission speed issue, is there any way to buffer the screen data and then update it all at once? The docs didn't mention anything like that, so I'm guessing no, but I might be wrong. I'm just hoping to find a way to get more responsiveness with this library.

ladyada, 2023-09-30:
yes someone could help write a variant of this library that can write the 4 data bits at once instead of one at a time, optimizations would help but of course would have to be implemented for each chipset

DutchComputerKid, 2024-12-17:
> I just bought the _I2C/SPI backpack_ (#292) with a _Standard LCD 16x_2 (#181) and am using it with a _MCP2221_ (#4471) on a PC running Debian 11. It takes over 6 seconds to write "Hello [...]

Yep, same here, all new hardware with a backpack, MC2221a and a 20x4 LCD. 
Planned to make a little status gadget for Linux servers or desktops but with the draw times this slow and needing to redraw the entire screen every time.... Anyone got a solution for this, or is the hardware just kinda 'useless'?

dhalbert, 2024-12-18:
#80 speeds up writes to the display. It's especially noticeable on slow I2C controllers like MCP2221.

closed 2024-12-18 by dhalbert