adafruit/Adafruit_CircuitPython_CharLCD #58: LCD with i2c backpack blinks but no text

paulbaumgarten, 2021-03-05:
Hello

I am having issues getting a 16x2 LCD display with an i2c backpack to work. When I power/start up, the LCD backlight blinks for about one second displaying no text, it then turns off for the 5 second sleep command, and then quickly blinks again for a fraction of a second.

The LCD works perfectly with my Arduino but not the M4. I have replaced the wiring. I have substituted for a different Metro M4 express board (I have a few on hand). No success.

I am using CircuitPython 6.1.0 and a freshly downloaded CircuitPython library bundle 6.x 20210303 with an Adafruit Metro M4 express board (ie: the purple ones).

My test code follows.

```python
import board
import busio
import time
import adafruit_character_lcd.character_lcd_i2c as character_lcd

i2c = busio.I2C(board.SCL, board.SDA)
lcd = character_lcd.Character_LCD_I2C(i2c, 16, 2, address=0x27)
lcd.display = True
lcd.backlight = True
lcd.message = "Hello\nCircuitPython!"
time.sleep(5)
lcd.clear()
```

Any advice is appreciated.

dhalbert, 2021-03-05:
Is it an Adafruit backpack? There are different kinds of I2C backpacks with different protocols.

paulbaumgarten, 2021-03-05:
Ah that would be it. It's a generic from the local electronic store. PCF8574T chip on the back. Are drivers available for that chip? I can't see it in the library bundle.

dhalbert, 2021-03-05:
I have a library I wrote. The API is different: https://github.com/dhalbert/CircuitPython_LCD. There are two branches available: `master` and `minimal`. The latter has a lot of functionality removed: I needed it to make it fit on a SAMD21.

paulbaumgarten, 2021-03-05:
Great! I'll take a look. Thank you for your help.

closed 2021-03-05 by paulbaumgarten