adafruit/Adafruit_CircuitPython_CharLCD #65: Could not display degree sign properly

daqingli616, 2021-09-06:
I tried to use your driver to display a string with degree sign in it. The string I tried to display is 

uString = 'Temp: 30' + u'\N{DEGREE SIGN}' + 'C'
print (uString) shows: Temp: 30°C
However, the following code only shows: Temp: 30-C
lcd.message = uString

It is run on a Raspberry PI4 with latest Raspbian and python 3.7.3.

larrycai, 2021-12-06:
there are two ways for degree sign

1. use default character in its RAM, actually it is `0xDF`, see https://mil.ufl.edu/3744/docs/lcdmanual/characterset.html, it may differ for different LCD, check its datasheet
2. create customized characters, code like https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/blob/main/examples/charlcd_customcharacter.py

FoamyGuy, 2024-11-07:
Closing this, the post above appears to provide a solution to the issue.

closed 2024-11-07 by FoamyGuy