adafruit/Adafruit_CircuitPython_CharLCD #67: While using the i2c/SPI LCD Backpack Maximum recursion depth exceeded

Beau28713, 2021-10-17:
While using the i2c/SPI LCD Backpack Maximum recursion depth exceeded

Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Metro M0 Express with samd21g18

Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "adafruit_character_lcd/character_lcd_i2c.py", line 31, in <module>
  File "adafruit_mcp230xx/mcp23008.py", line 16, in <module>
  File "adafruit_mcp230xx/mcp230xx.py", line 16, in <module>
  File "adafruit_mcp230xx/mcp23xxx.py", line 14, in <module>
  File "adafruit_bus_device/i2c_device.py", line 14, in <module>
RuntimeError: maximum recursion depth exceeded

```python

import time
import board
import busio
import adafruit_character_lcd.character_lcd_i2c as character_lcd

lcd_columns = 20
lcd_rows = 4

i2c = busio.I2C(board.SCL, board.SDA)

lcd = character_lcd.Character_LCD_I2C(i2c, lcd_columns, lcd_rows)`

mmorys, 2021-11-22:
I just experienced the same issue:
CircuitPython 7.1.0 Beta 0; Seeeduino Xiao with SAMD21G18 microcontroller.

Using a 16x2 monochrome LCD with integrated PCF8574T I2C connection.

xgpt, 2022-01-26:
Also experiencing this issue.

stonehippo, 2022-08-31:
This is definitely an issue with the limitations of the M0/SAMD21 boards. I ran into this using an M0 Express, so I tried it with an SAMD51 board (Grand Central M4) and things worked great. The memory available to the M0 is just not large to import the underlying mcp230xx driver. You can see this in the REPL with:

```
>>> from adafruit_mcp230xx.mcp23008 import MCP23008
MemoryError: memory allocation failed, allocating 640 bytes
>>>
```
Importing that module is the first thing the I2C implementation does. The same is true when using SPI and the 74hc595 driver.

The answer is to use an M4 or another, more capable board to drive this backpack.

dhalbert, 2024-12-18:
Closing for now.

closed 2024-12-18 by dhalbert