adafruit/Adafruit_CircuitPython_CharLCD #78: ValueError: incompatible .mpy file

SerialDesignatorN, 2024-01-28:
I tried using charlcd_i2c class in my circuitpython code, and all it says is just incompatible .mpy file. I tried all solutions, including rolling back to an older bundle from 6.x, and none of that seems to work.

dhalbert, 2024-01-28:
https://circuitpython.org/libraries explains in detail about versions. You need the versions of the libraries to match up with the version of CircuitPython.

1. Upgrade your CircuitPython to 8.2.9. See https://circuitpython.org/downloads to download.
2. Record which libraries are in the `lib/`. Then remove them and install libraries that match your version of CircuitPython.
3.  Make sure also that there aren't .mpy files in the top level of CIRCUITPY. They will be used before versions that are in `lib/`.

FoamyGuy, 2024-11-07:
Closing this. Please post on forums or the help-with-circuitpython channel in discord https://adafru.it/discord if your still having trouble.

closed 2024-11-07 by FoamyGuy